uob-hep-cad/uob-hep-pc072 #18: Remove D-FF from upstream SFPs

DavidCussans, 2023-12-13:
The three upstream SFPs can each be connected to a different timing source. Hence return data can't be clocked from same clock.

--> Remove D-FF (driven by the single board PLL) registering the data going from FPGA to SFP Tx

DavidCussans, 2024-06-21:
> Sorry I think I might need a bit of help again here. I know I scribbled something down when we spoke but I can't find the paper. Devices U3/U8 need to be removed, leave U2 for SFP<2>/third SFP.
>
> The newly placed IC3 (854S01) part is to replace "U2" for SFP<2>??

Pete,

Yes, the “issue” doesn’t give much detail.

On two upstream channels remove the D-FF on SFP 0,1 and connect the FPGA directly to the SFP Tx pins:

U8 should be removed and UPSTREAM_DATA_{P,N}(0) connected directly to SFP_TD_{P,N}(0) via C106/C100

U3 should be removed and UPSTREAM_DATA_{P,N}(1) connected directly to SFP_TD_{P,N}(1) via C101/C98

On the third upstream channel it should be possible to switch between direct connect to the FPGA and connection via a D-FF:

SFP_TD_{P,N}(2) should be disconnected from U2 and instead connected to the output of multiplexer IC8 ( 854S01 )

One input of IC8 should be connected to the output of U2

For the other input of IC8 there are two choices:

  1.  Disconnect one pair of AUXIO signals from J4 and instead connect to U8 ( adding DC-blocking capacitors in series and adding 100-Ohm termination resistor near U8)
  2.  Disconnect UPSTREAM_DATA_{P,N}(2) from U2 and connect to the input of a 1:2 LVDS fanout ( Use a SY58608 like IC2 etc. ). One output of the newly added 1:2 fanout should go to the input of U2. The other output should go directly to the second input of U8

I don’t have a preference for which option is chosen ( 1. or 2. ). Go with the lowest staff cost one. Probably option 1. I guess.

Cheers!
   David

oxfpete, 2024-06-27:
Think it is almost done, need your eyes to look at schematic.

DavidCussans, 2024-07-03:
Schematic change looks good.
Need to double-check that it is OK to connect the "direct out" signal to B16_L14_P/N

DavidCussans, 2024-07-23:
Need to check that direct output is on suitable FPGA pins.

DavidCussans, 2024-09-11:
Both clocked and direct output data pins are on same bank (16)

Outputs are on the same bank as input

DavidCussans, 2024-09-12:
Closed after review

closed 2024-09-12 by DavidCussans