uob-hep-cad/uob-hep-pc072 #19: Allow AVR programming cable to override RESET line from uTCA backplane

DavidCussans, 2023-12-13:
When MIB unplugged from backplane RESET line is asserted. This prevents correct programing of MMC uController (ATMEGA128) via JTAG.

---> Add circuitry that allows programming cable to override RESET line

oxfpete, 2024-06-19:
Done.

DavidCussans, 2024-07-03:
Schematic looks good.

DavidCussans, 2024-07-22:
PL1 needs to move. The space between the two mounting holes is taken up with a solid block of plastic that stabilized the joint between the main PCB and the tongue-2 PCB

oxfpete, 2024-07-23:
PL1 moved closer to C58

oxfpete, 2024-07-24:
PL1 well away from C58 and original position, above SK1 and clear of Trenz module

DavidCussans, 2024-07-24:
Looks Good To Me

Dr David Cussans                    University of Bristol,
office: +44-(117)-95 46879   H.H. Wills Physics Laboratory,
     lab: +44-(117)-33 17199   Tyndall Avenue,
     fax: +44-(117)-925 5624   Bristol BS8 1TL, UK

DavidCussans, 2024-09-12:
Problem spotted - this allows us to program on bench,
but Stoyan points out that this won't allow us to program in crate - since MCH will apply reset at intervals

Conclusion - Pass for pre-production. Continue investigation

closed 2024-09-12 by DavidCussans