uob-hep-cad/uob-hep-pc072 #22: Add fuses to main 12V and 3.3V management power line from backplane

DavidCussans, 2023-12-15:
Fermilab electrical safety reportedly insist on fuses at power input to boards. This isn't usually done on uTCA boards, but add fuses to make life easier for ourselves 

Normal operating power for MIB ~ 12W ( i.e. 1A from 12V )

Put in 3A slow-blow surface mount fuse.

oxfpete, 2024-06-21:
SF-0603FP015F-2 will used on 3V3 line, it is a 150mA fast fuse - hopefully has enough head room to compensate for in rush at power on. 100mA was originally requested but not available in slow-blow as an 0603.

oxfpete, 2024-06-21:
Done.

DavidCussans, 2024-07-03:
Schematic looks good.

DavidCussans, 2024-07-23:
Measure plastic blocks between main PCB and tongue-2 and make sure that FS2 doesn't interfere

oxfpete, 2024-07-24:
F2 and F1 moved well away from plastic block locations

oxfpete, 2024-07-24:
R11 may still be close - please check

DavidCussans, 2024-07-24:
R11 is OK. It doesn't get squashed under the plastic block on the current design so doesn't need moving.

( The block is 6mm wide (+/- 3mm) , and R11is 4.36mm below the mounting holes )

Dr David Cussans                    University of Bristol,
office: +44-(117)-95 46879   H.H. Wills Physics Laboratory,
     lab: +44-(117)-33 17199   Tyndall Avenue,
     fax: +44-(117)-925 5624   Bristol BS8 1TL, UK

DavidCussans, 2024-09-12:
Not necessary since FNAL safety office have accepoted that uTCA baords don't need fuse.

However, reduces QC requirements on MMC code, so leave in.

Passed after review