uob-hep-cad/uob-hep-pc072 #41: One mounting block for tongue-2 PCB needs to have recess milled to avoid C188/C189

DavidCussans, 2024-07-22:
There are plastic blocks that fit between the main MIB PCB and the tongue-2 PCB
( between MH7 and MH8 , MH9 and MH 10 ).

One the the plastic blocks covers C188/C189. Moving the caps seems like a PITA, so remember to add a milled recess into the CAD design for the blocks.

(N.B. - Get the CAD drawing from Keith and add to the repository.)

oxfpete, 2024-07-24:
Caps moved to back side, plastic blocks should both be clear of components

DavidCussans, 2024-07-24:
Looks good.

Dr David Cussans                    University of Bristol,
office: +44-(117)-95 46879   H.H. Wills Physics Laboratory,
     lab: +44-(117)-33 17199   Tyndall Avenue,
     fax: +44-(117)-925 5624   Bristol BS8 1TL, UK

DavidCussans, 2024-09-12:
Closed - DGC checked last night

closed 2024-09-12 by DavidCussans