uob-hep-cad/uob-hep-pc072 #44: Move  blue LED to correct position to follow uTCA standard.

DavidCussans, 2024-07-23:


oxfpete, 2024-07-24:
I don't think the new position is correct at all on the layout I have just submitted. Is that mounting point on the PCB I have?

DavidCussans, 2024-07-24:
The blue LED is in roughly the right place - on the bottom near the front of the board - but needs tweaking.

The image at https://github.com/user-attachments/assets/12ca9d7b-fdcc-4e8c-895e-896952bf6baa shows the position relative to MH6 on the DUNE MIB. ( There are other holes on the AFC which are used to mount the FMCs but these aren't present on the MIB ).

In its new position LED also needs to be a "side launch" (like on the FIB ) to type to shine the light horizontally (rather than a vertical mount )

Dr David Cussans                    University of Bristol,
office: +44-(117)-95 46879   H.H. Wills Physics Laboratory,
     lab: +44-(117)-33 17199   Tyndall Avenue,
     fax: +44-(117)-925 5624   Bristol BS8 1TL, UK

DavidCussans, 2024-08-28:
Blue LED , DS1 still needs to be moved.

oxfpete, 2024-09-02:
Done

DavidCussans, 2024-09-11:
Checked on mib_rev3ox_v223.

Looks Good To Me

closed 2024-09-12 by DavidCussans