uob-hep-cad/uob-hep-pc072 #45: Check to see if data from AFCs arrives on clock-capable pins on FPGA.

DavidCussans, 2024-07-23:
Would give ability to measure fine-phase of returning data.
May be difficult to implement.

DavidCussans, 2024-07-31:
May not be needed. Phase measurement blocks use BUFR which can be driven by normal I/O

closed 2024-09-12 by DavidCussans

strilov, 2024-09-17:
CC pins with non-clock input

bank 13:

```
Y11   IO_L11P_T1_SRCC_13
Y12   IO_L11N_T1_SRCC_13
```

```
W11   IO_L12P_T1_MRCC_13
W12   IO_L12N_T1_MRCC_13
```

```
U15   IO_L14P_T2_SRCC_13
V15   IO_L14N_T2_SRCC_13
```

bank 14:
```
U20   IO_L11P_T1_SRCC_14
V20   IO_L11N_T1_SRCC_14
```

```
W19   IO_L12P_T1_MRCC_14
W20   IO_L12N_T1_MRCC_14
```

```
Y18   IO_L13P_T2_MRCC_14
Y19   IO_L13N_T2_MRCC_14
```

```
V18   IO_L14P_T2_SRCC_14
V19   IO_L14N_T2_SRCC_14
```

bank 15:
```
J20   IO_L11P_T1_SRCC_15
J21   IO_L11N_T1_SRCC_15
```

```
K18   IO_L13P_T2_MRCC_15
K19   IO_L13N_T2_MRCC_15
```

bank 16:
```
B17   IO_L11P_T1_SRCC_16
B18   IO_L11N_T1_SRCC_16
```

```
E19   IO_L14P_T2_SRCC_16
D19   IO_L14N_T2_SRCC_16
```

CC pins with clock input (not available)

```
data_from_upstream_0
V13   IO_L13P_T2_MRCC_13
V14   IO_L13N_T2_MRCC_13
```

```
data_from_upstream_1
L19   IO_L14P_T2_SRCC_15
L20   IO_L14N_T2_SRCC_15
```

```
data_from_upstream_2
D17   IO_L12P_T1_MRCC_16
C17   IO_L12N_T1_MRCC_16
```

```
CLK_GEN_IN_1
J19   IO_L12P_T1_MRCC_15
H19   IO_L12N_T1_MRCC_15
```

```
CLK_GEN_IN_0
C18   IO_L13P_T2_MRCC_16
C19   IO_L13N_T2_MRCC_16
```

strilov, 2024-09-29:
One option to get 10/12 BP data links into CC pins, with some caveats:

* Some PLL/SFP controls signals move to 2.5V IO (from 3.3V on bank 14). 2.5V CMOS output levels should be compatible with 3.3V CMOS input levels.
* Artix 7 does not support LVDS_33, and bank 14 is fixed at 3.3V. Artix 7 does support the LVDS25 inputs into IO bank at a different voltage, but internal termination cannot be used -> external termination required.

```
swaps to free up B14_L11 
SFP_TX_DISABLE_0 : B14_L8_N -> B15_L18_P (output 2.5V CMOS into 3.3V input on SFP) 
SFP_TX_FAULT_0 : B14_L8_P -> B14_L24_P 
SFP_TX_FAULT/LOS 2 : B14_L11 -> B14_L8
BP_DATA_IN_3: B16_L24 -> B14_L11 (LVDS25 into 3.3V IO bank, will require external termination)
```

```
swaps to free up B14_L12
CLK_SEL: B14_L10 -> B15_L23 (output 2.5V into 3.3V input on PLL)
CLK_GEN INTR/LOL : B14_L12 -> B14_L10
BP_DATA_IN_9: B16_L16 -> B14_L12 (LVDS25 into 3.3V IO bank, will require external termination)
```

```
swaps to free up B14_L14
SFP_TX_DISABLE_1 : B14_L17_P -> B15_L16_P (output 2.5V into 3.3V input on SFP) 
SFP_LOS_1 : B14_L14_P -> B14_L17_P
SFP_TX_DISABLE_2 : B14_L14_N -> B15_L16_N (output 2.5V into 3.3V input on SFP)
BP_DATA_IN_8: B16_L19 -> B14_L14 (LVDS25 into 3.3V IO bank, will require external termination)
```

```
BP_DATA_IN_0: B13_L3 -> B13_L14
```

```
BP_DATA_OUT_5 : B13_L11 -> B13_L9
BP_DATA_IN_1: B13_L9 -> B13_L11
```

```
BP_DATA_IN_4: B16_L23 -> B15_L13
```

```
BP_DATA_OUT_4: B15_L11 -> B15_L3
BP_DATA_IN_5: B16_L20 -> B15_L11
```

```
DIRECT_US_DATA_2:  B16_L14 -> B16_L10
BP_DATA_IN_7: B16_L18 -> B16_L14
```

```
BP_DATA_IN_6: B16_L9 -> B16_L11
```

```
AUXIO_2: B13_L12 -> B13_L2
BP_DATA_IN_2: B16_L22  -> B13_L12
```