uob-hep-cad/uob-hep-pc072 #47: Move red and green LEDs to "standard" positions for Schroff/nvent front panel mechanism. Remove existing light-pipe and rely on front panel mechanism

DavidCussans, 2024-07-31:
Schroff/nVent front panel mechanism includes light pipes for the two auxilliary LEDs ( red / green )
( https://schroff.nvent.com/en-gb/products/enc20849-153 )

--> Remove existing light-pipe

--> Move Red and Green LEDs to correct positions to suit nVent light pipe
( https://schroff.nvent.com/sites/g/files/hdkjer281/files/acquiadam_assets/2021-06/70849-045.pdf )

N.B.
1. Can probably measure positions from AFC
2. The LED nearest front of board should be replaced with an "edge launch" . The one furthest away should stay as vertical launch.

DavidCussans, 2024-08-28:
mob_rev3ox_v221.brd looks like LED positions incorrect.

on LED D1 with respect to mounting hole: delta-x = 6.97mm , delta-y = 1.1mm ,

looking at AFC design it looks like this should be delta-x = 5.583mm , delta-y = 1.069mm . i.e. y position OK within tolerance, x position needs to be shifted by ~ 1.4mm

Screenshot from Altium: [image]

oxfpete, 2024-09-02:
LED D1 moved.

DavidCussans, 2024-09-12:
Closing since
1. Checked by DGC
2. WOn't stop the board working

closed 2024-09-12 by DavidCussans